spacedecentral/Coral #1: Compile research

Engineer1119, 2018-07-16:
- Lunar resources
- Moon-focused commercial efforts
- Moon-focused governmental efforts
- Lunar environment

founder-outbound, 2018-07-31:
I have some experience in this, and would be happy to help.

Suzibianco, 2018-07-31:
We are just beginning our repo so there might not be a lot to do yet. 
For now, please share your views on the current organization we have for the tasks, and how we can make it better!

timallard, 2018-07-31:
I liked the layout & categories, I'm ok at coming up with tasks for workflows in windows, team foundation server admin a lot in dev, know project style software.
Github acct ok, riot installed, will get used to things, topics and tasks ...

cagonzalez8, 2018-09-06:
@Suzibianco @Engineer1119 @stellarmagnet @capsulecorplab @founder-outbound I mentioned in the meeting about helping out with the organization (possibly in the form of a Gantt chart). I'm not going to lie, I'm having a hard time keeping up with everything. Is it cool with y'all if I can get some time to catch up to see the WHOLE picture so I can organize my notes?

Suzibianco, 2018-09-06:
Although Monday is not good for @capsulecorplab. Sean, when do you propose we do this?

cagonzalez8, 2018-09-06:
I vote for tonight or Friday night, preferably before party time.

timallard, 2018-09-07:
Seeing the library index this sheet mixes that with subsystem & other dropdowns on a form to enter instead of directly into a sheet. I pasted in the library entries: https://docs.google.com/spreadsheets/d/19Dxx1TR8TzuG8V7b1kccRNZXauwMcpIxtGw2afHLcgI/edit?usp=drivesdk&ouid=111702438340332674503

Engineer1119, 2018-09-08:
@cagonzalez8 It looks like Suzi and Sean are occupied, and unfortunately I got home from work a short while ago, so you'll have to endure party time without the certainty of a gantt chart in the back of your mind. Would sometime Sunday work for you? [This](https://docs.google.com/drawings/d/1SRn0RPTPhZIMpwhXUDYXaHQod2ruh5NN_CuBsmpXvD0/edit?usp=sharing) is the very rough diagram I mentioned in-meeting this past Monday.

founder-outbound, 2018-09-08:
Looks like you put some significant effort into this. Good Job! I'll review it.-Craig

cagonzalez8, 2018-09-08:
@Engineer1119 Sunday's fine. What time? And thx for the diagram.

Engineer1119, 2018-09-09:
My best guess is after 3pm Pacific time, how about you?

timallard, 2018-09-11:
Got the g.sheets way to do dropdowns, this has wider scope on categories: https://docs.google.com/spreadsheets/d/1UXbiBrxpAuVqbPg4N5lt7cKDxAPAL6K63pvjyUUMte4/edit?usp=drivesdk

cagonzalez8, 2018-09-13:
Oh man, I'm just finally seeing these. Sorry about that. But on a housekeeping note, I started a rough Gantt chart on my Hansoft server, and essentially what I'll be doing is making a parallel to it and the general tasks set here on Github. Would that be safe execution to assume?
> 
> My best guess is after 3pm Pacific time, how about you?

Engineer1119, 2019-01-20:
This is strongly related to issue #84 to the point where it's borderline deprecated.  I'll limit the scope of this issue to issues 1 through 100 and consider it closed.

closed 2019-01-20 by Engineer1119