spacedecentral/Coral #21: Resource Characterization

Engineer1119, 2018-07-24:


Engineer1119, 2018-09-02:
[Here ](https://docs.google.com/spreadsheets/d/1bkHw_ekMEbXps_7yKP-LLcbINslKgCAeYVVvDBRmCXI/edit#gid=0) is a partial database of material properties of interest

timallard, 2018-09-13:
For elemental iron I've only found reference to "microscopic" particles no density/volume figures yet on mare soils.

jrcgarry, 2018-09-13:
<nods>
My understanding is that its thought to be present (primarily) as a nanophase, condensed from meteoritic impact processes.
<modest justification for that belief here>

Page 3:
http://www.isruinfo.com/docs/the_lunar_dust_problem_-_from_liability_to_asset.pdf

Suzibianco, 2018-11-01:
[Closing this issue as we migrate to a new phase in Coral and new process of how we document issues.]

closed 2018-11-01 by Suzibianco