spacedecentral/Coral #26: Assembly Hardware

Suzibianco, 2018-07-24:


founder-outbound, 2018-07-31:
I have an intense desire to help on this task. Design is my baseline skill.

Suzibianco, 2018-07-31:
Hey Craig, this is great!
This task is still in a definition stage, and will be part of a working group later on. But you can begin researching! Please open up a subfolder in the google drive [TBO](https://drive.google.com/drive/folders/10P_UcQ34iZ-9PBl6CWYXjORbEdqsWKvB?ogsrc=32) folder for your findings. Once we have the working groups sorted out, we will know where to store it permanently.

Suzibianco, 2018-09-05:
Hey @founder-outbound, I'm closing this task because we won't really be doing the structural assembly. But you can still work on the assembly of the system, printer, etc right?

founder-outbound, 2018-09-06:
Certainly!

closed 2018-09-06 by Suzibianco