spacedecentral/Coral #28: Electrical Power

Engineer1119, 2018-07-24:


timallard, 2018-07-31:
Inverters aren't very long lived went looking and found DC to AC converters used before solid-state and they are direct drive, so, by giving the motor a longer torquearm it gains mechanical advantage on the generator, a class 2 lever 4:1 feasible.
I call these feedback generators for a class of machine, the motor rotational velocity is higher cutting more lines/sec than the generator if the same turns/unit a way to view them, 5.125kw can output 4kw, motive power or electricity.
Proof of concept can be done with drone parts, matching rpm the main trick, haven't done this yet no place to work.
In spite of that I have designs from palm sized phone chargers to 125kw using magnetics, in space they don't use fuels, repolarize every 30+years 2-3 times.
Before this my focus using these was firefighting aerobot tankers, drone fire nets with controller drones ... these could be a big deal, hope you're interested, for veracity it needs eyes, cheers, tom

Suzibianco, 2018-08-01:
@timallard Do you have literature research about this, or examples of your own experience? We want to get a good set of references to base our selection of power systems on.

timallard, 2018-08-01:
Suzibianco,

stellarmagnet, 2018-08-11:
Hey @timallard since you were starting the power research can you please follow these instructions for the task?

timallard, 2018-08-12:
Will do, have an outline of most mechanical kinds for high temp & thermal goes with that, we are saved by solar-direct no atmosphere for high temp, standard fare on the rest plus feedback motors, a self-powered motor-generator. Quantifying these to kg's is like ......... oooo kay ... I want to survey to get brands in-use, their product sheets may have it for aerospace, nasa files may have leads, cheers!

timallard, 2018-08-12:
This is a draft outline of power venues, first pass: https://docs.google.com/spreadsheets/d/17Zd_kLVw36AeLOeYqf_eo0RuOpanrBA7BZoO32wb97M/edit?usp=drivesdk

timallard, 2018-08-12:
This is the equation for rating motor efficiency by weight any kind that seems valid, standard machines rarely vary from 1:1 but they, so, need to be measured to the air gap as the torquearm lengths matter, without that it's a guess, feedback motors are motor-generator units with 4:1 mechanical advantage, load torque radius needs to match the generator for rating, can be more or less by less or more arm length: 

Power-Weight Factor, P-W ≈ ((P.out)(Driver/Driven)(1-%loss/watt))/Weight
Standard 1:1; Feedback 4:1 Driver/Driven, must be on the same shaft, direct drive equivalent and match rpm range reasonably well.

Suzibianco, 2018-08-19:
Hey @timallard we see you putting a lot of effort into creating these sheets and that is awesome - however, the objective of this task is to conduct literature review on the different power technologies, and document them according to a few factors, and since this is a collaborative task, we need a standard list of factors to compare. That's why we already created a spreadsheet and would like you (and the other people interested on this task) to populate the spreadsheet with the various technologies.  This is the link for the spreadsheet: [https://docs.google.com/spreadsheets/d/1-EBZgltVHDXnP13Wnn71i3uRfV2lRPFsy1u-j3tIBFg/edit#gid=0](https://docs.google.com/spreadsheets/d/1-EBZgltVHDXnP13Wnn71i3uRfV2lRPFsy1u-j3tIBFg/edit#gid=0 )

We need to keep things organized, since we have 25 people in the team, so we would like to have standard procedures for how we complete the tasks. This particular task is for literature review only - if you are interesting in producing templates for our spreadsheets, documents, trade studies, etc, please refer to task #72 and #73 

Thanks!

timallard, 2018-08-20:
Found Momento db android, it allows user entries, the form for the power-weight ratio by supplier is fast, brands search later, db can be scripted! Uses 2 github to even allow connections to the internet.

Links directly to sheets with results, test page going here: https://docs.google.com/spreadsheets/d/1umfPoDMF68BN8er-EnqDyhhcNhZb-reL69VF5xyk4mY/edit?usp=drivesdk&ouid=111702438340332674503

timallard, 2018-08-21:
Revised, auto-fill added on brands: https://docs.google.com/spreadsheets/d/1-7j1I6n29Xh5T5HrXi6LqklCcqddJp2YO2buL9wT4II/edit?usp=drivesdk&ouid=111702438340332674503

timallard, 2018-08-21:
Revised, added columns & eq. for unit volume & pwr-wt-vol ratio: 
https://docs.google.com/spreadsheets/d/1F3ZwGTFH2v6XNhTrLp9ZWvHxqZGnSv4UDetPobjygfk/edit?usp=drivesdk&ouid=111702438340332674503

timallard, 2018-08-21:
I found a terse bibliography tool that categorizes , organizes well a companion to the db tool, looks fitting for a literature review, Bolshoi on ggle play, survey tool is Memento DB that has all the options for teams, both share to the std list includes docs, sheets & drive.

timallard, 2018-09-04:
This is the best so far of useful info one stop, it has over 20 methods of non-destructive testing, what I was searching for, with many other aerospace & industrial specific topucs, latest stuff, by subscription:  Mechanical Engineering App on Google Play, To download https://goo.gl/Jfn3Qx

timallard, 2018-09-05:
Adding the library index headings with subheadings I did for the form dropdowns.

timallard, 2018-09-06:
Finished an article on permagnetic & emf feedback motors that emulate the permanent magnet version, drawings: https://drive.google.com/file/d/1mghYRn4AVItmY2NL5Xbdub7yIGTgtm0_/view

Engineer1119, 2018-09-09:
Thanks for putting together some documents Tom. A couple questions:
- First, the engineering-side question that is more important to Coral. Marc asked if you knew of any documented research or demonstrations of the system you outline. Do you think you'll be able to find some for him? This question is key from a pragmatic standpoint: Coral is already intended to push the TRL in several areas, and being the cutting-edge on too many fronts is difficult and risky in a lot of ways that we can go into if you want. The system you're proposing is the only one proposed so far with no history of terrestrial or space-based demonstration, and without more compelling evidence I have to recommend that you focus your efforts into looking at other areas within Coral. Please let me know if that makes sense.
- And the physics-side question to satisfy my personal curiosity. As you mentioned in-meeting, the ideas that a permanent magnet can create mechanical torque and that lever arms create mechanical advantage are high school physics. However, the proposed system (in the engineering sense of the word) would come to an equilibrium. E.g., a magnetic plumb bob would come to rest at some constant distance away from a permanent magnet placed directly below the point that it hangs from. If the plumb bob hangs from a rigid rod with a second permanent magnet attached further up the rod, then the equilibrium state is different but is still fundamentally equilibrium. What I can't tell from your linked documents is where the energy (Joules) comes from to consistently create torque (do "work" in the physics sense of the word, e.g. Newtons * meters). It takes energy to do work, and if the proposed system is used to do work (e.g. turn a rover wheel) then it is, by definition, expending energy.

timallard, 2018-09-13:
I got onto this from DC-AC converters that lasted years over solid-state inverters, reliability was a primary goal.
...Then, by only focusing on several machines we can sanction them for flight using permanent magnet motors of two types, one thermal the easiest, the other for electricity.
...By removing thermal from power needs it greatly reduces battery to make 2-weeks, Peregrine is at 2.5w continuous on 840-wh an example of no change to electrical, add in thermal to it.
...For wattage in general, a permanent magnet rotor compacts and lightens the machine they aren't too bad on control this size so the focus genset geometry.
...The closest OTS are axial rotors used in twin engine aircraft now, still not self-powered yet similar geometry, all can be modelled well, it's all ben done that way.
...I took Peregrine's solar at 540w for such a genset as a production goal by a machine with counter-rotating rotors for gyroscopic balance the design need.
...If it was newer tech I'd have little confidence to present it for a fast-track to be reliable with the prize of being self-powered, mmf or emf, the risk isn't unrealistic as a shop project for small machines.
... The coinmaker is a lot less a sure thing from any angle and it looks looks ok so far.
... As for work, magnets are pure N-m/lb-f once polarized, the constant field force is doing the work, it doesn't need electricity in the equation anywhere, that's the simplicity and 30+ years of operation for a rotor where one is aiming the stators at 45° to create the motive power, a loss of 1.4 on flux to gain the motion. With samarium cobalts they aren't at all heavy on 1- 2mm gaps so it's only downer is lack of control, I add emf for that with coil spring starts, they are best used for baseload generation any scale, for space power, to me, hybrids are a perfect fit, on earth not so critical and feedback emf machines emulate the permanent style these are called permagnetics with about 1/3 "loss" by adding more coils to the system.
... I don't have a shop or office atm, trying for forest land with a cabin for helping with sim & solidworks to pull this off to TRLs high enough, they're just very simple vs horizontal armatures.

timallard, 2018-09-13:
Regards what I've been doing for Coral, these spreadsheets are almost ready, the first for any electric motor or generator, the other called mash_all a try at a centralized posting place, almost no required fields with task fields, then landers:
https://docs.google.com/spreadsheets/d/172q3LLKLMIPZlmx9AMrvysNnOqlK2b_2jnf2a9CQjig/edit?usp=drivesdk&ouid=111702438340332674503
mash:
https://docs.google.com/spreadsheets/d/1UXbiBrxpAuVqbPg4N5lt7cKDxAPAL6K63pvjyUUMte4/edit?usp=drivesdk&ouid=111702438340332674503
landers is new:
https://docs.google.com/spreadsheets/d/1EEb7qjVUMn_wUbUidZ_rNdaa3ZmwI2yjDmY7jKb_3bs/edit?usp=drivesdk&ouid=111702438340332674503

timallard, 2018-09-26:
As a methodology to estimate the longevity of rare-earth, so-called permanent, magnets the question of that on physics stackexchange is 1% fade a decade, 10% loss per Century.

And significantly this is not reliable.

Thus my work has been to apply empirically derived metrics to quantify in terms of the magnetization power, temperature, binder and so on affecting the outcome of the polarization process which is done near solidus temperature where the molecules align easily and can be "frozen" in that structural arrangement to last so long and provide non-natural, elevated flux-densities not available before the advent of rate-earths able to use #repulsion and thus the first practical forms to apply the purely magnetic force to motors.

I challenge all other views to accurately assess longevity at this time.

timallard, 2018-09-27:
> 
> 
> Tom,
> If I have a permanent magnet, it has a magnetic field that does not change with time.
> 
> > the constant field force is doing the work
> 
> I am looking at a magnet on my desk right now.
> What work is it doing?
> 
> (If I move a charge past this magnet, then 'yes' I do work to counteract the Lorentz force)
> 
> I am genuinely none-the-wiser.
Say you have a rotor about 4cm in radius with all N one side S the other and angle 100 magnets as stators into that at 45-degrees you'll gain motive power.

This is using axial format and stators are designed to only accelerate, it's not a phased machine, there is a constant flux rate in rare earth magnets that can't show loss in decades or it's a bad batch so for a moon shot of three years the mag's can be paper thin, just enough to supply the flux density to spec.

The longevity to me is now at more than 65-years of steady power before the decay cycle begins, doing a graph of the longevity cycle of them so people get they don't work like batteries, they store potential energy and can only emit a fixed amount, when energy drops below what transmits full power the excess PE is gone and de-polarization begins.

So, that's how and where the large amount of useful magnet-to-magnet power comes from, there are no other inputs and the energy is "stored" by preventing molecules from returning to their base state mainly via the high density of cobalts, 8.9 ... when blobs elongate in the high flux they can't physically move anymore and loss power from loss of polarity & magnetic moment only.

timallard, 2018-09-29:
From all views I can see the only choice of typical power sources is to work with Peregrine to put another of their panel-battery combos on the lander if flight stability handles it, there are no alternatives I can see that can fly in 3-5 years.

timallard, 2018-10-03:
I'm replying to this question, First, the engineering-side question that is more important to Coral. Marc asked if you knew of any documented research or demonstrations of the system you know of outline, I ask: "What the specific geophysical and physics answer simple to: Ferromagnetics are a man-made magnet from a material specific to magnetic flux retention upon quenching after magnetizing and polarization methods upon modern "rare earth" materials, if anyone doubts this Mercedes Benz electric motor rotors ere identical, go review your intellect your trite arguments are just that.