spacedecentral/Coral #33: Create a database for papers on ISRU and their possible end users

Suzibianco, 2018-07-30:


founder-outbound, 2018-07-31:
I would be happy to collect and organize this information.

Suzibianco, 2018-07-31:
Great Craig! This is an organization task, dependent on the research task "[Literature review on key Coral topics: Moon, regolith, 3D printing](https://github.com/spacedecentral/coral/issues/36)". So it might take a while before there's much to organize.
You can begin by checking out our [proto-wiki](https://drive.google.com/drive/folders/1AB5cbHr8RqisjpuU8uWroTczjzrJU5Yj?ogsrc=32) to sort the documents we already have in there. Please save this spreadsheet in the Wiki folder.

Suzibianco, 2018-08-17:
@founder-outbound, as we talked last week, this task is no longer needed because each person who researches something will be responsible for documenting that research in a spreadsheet.

closed 2018-08-17 by Suzibianco