spacedecentral/Coral #41: Thoughts on the Coral mission

Suzibianco, 2018-07-30:
Please create a subtask for your answer, and add your name to the title. Once done, move it to the "Review" column.

Write a 2-3 paragraph essay on your thoughts in relation to Coral. try to answer the following:

- What do you envision when you think about the Coral Mission?
- What are some of your expectations for this mission?
- How do you think we will achieve the long-term objective of building a habitat on the moon?
- If you had to describe a concept for the entire mission right now, what would it be?

timallard, 2018-08-01:
Two short pages, underground living, to deal with dust rails for transport, using algae purification of wastewater to remove CO2, produce oxygen, use the oils and algae mass for products, i.e. lubes, soaps.
Recalling Biosphere 3 for the need tovl remove CO2, sequestration isn't enough.
https://drive.google.com/folderview?id=16oAE-1rn5a8DdtFp4nbwP1Mx8kKXXNnY

Suzibianco, 2018-08-01:
Awesome @timallard, can you put this in a folder with your name in our shared drive, more specifically [here](https://drive.google.com/drive/folders/1hrcON9dMOrLtYwJhgslnBvT7cnDvPPce?usp=sharing)?

closed 2018-08-22 by Suzibianco