spacedecentral/Coral #43: Provide your thoughts on the Coral mission - Tom Mallard

Suzibianco, 2018-08-01:
@timallard Please add the documents you shared with me to this folder: https://drive.google.com/drive/folders/1hrcON9dMOrLtYwJhgslnBvT7cnDvPPce?usp=sharing

timallard, 2018-08-02:
[image]
[image]

Suzi, let me know if these don't work, had a chore getting them to devices.

Suzibianco, 2018-08-02:
Hey @timallard, I meant put them in the google drive folder: https://drive.google.com/drive/folders/1hrcON9dMOrLtYwJhgslnBvT7cnDvPPce?usp=sharing

They did show up hete though! And thank you for the illustration concept!

timallard, 2018-08-04:
I forgot, algae oils can be used as crude oil now to supply plastics and other products.

timallard, 2018-08-04:
Excavation, the dirt likely has horizontal very fine layers of ejecta events, impact or volcanic the thought to focus collection on these as less volume to process, i.e. sensing metals for tool guidance the idea first, overturning overburden vs digging for gauging total volume moved for it.

timallard, 2018-08-16:
New vision: Satisfy all requirements in demo if not needed to produce pigmented glass cloisonnais 3D coins microprinted with the patch design as product: https://drive.google.com/file/d/1Y2x67w-WPNFNW6UYFpsou9sGDCvCGseU/view?usp=drivesdk

closed 2018-08-17 by Suzibianco