spacedecentral/Coral #44: Design logo for Coral mission

Suzibianco, 2018-08-02:
Design a mission logo for Coral. It can be a logo or a mission patch, but it has to be legible even when it is very small. Please use only open source/public domain images, or your own. 
Submit a png file for approval. If you have questions about the design, please comment here.

cagonzalez8, 2018-09-26:
I want something like this.
[image]

Suzibianco, 2018-09-28:
Hell yeah! Hahahaha.

cagonzalez8, 2018-09-29:
Whenever it actually happens, I'll play around with a couple main logo and a mission patch design(s) when I got some free time to kill.

jrcgarry, 2018-09-29:
My attempt here:
(Inkscape, so all vector: and drawn while waiting for Friday rush hour to subside)
[image]

Suzibianco, 2018-10-01:
@jrcgarry I'm not sure what I'm looking at here, is this a cross between the moon and a coral? Nice try though!

jrcgarry, 2018-10-01:
<sad nod>
Not exactly inspired is it?

-JG
(keeping the day job)

cagonzalez8, 2018-10-01:
I like it. needs some color.

jrcgarry, 2018-10-12:
I described the project to an artist - this is what she made yesterday.
<liking the blue>
[image]

ggabogarcia, 2018-10-28:
Hello, I would like to contribute with these topics, that is why I designed this logo, for its simple forms, it works in small sizes. Any suggestions and comments are welcome.
# 
[image]

Suzibianco, 2018-10-28:
Hi @ggabogarcia, I'm glad you want to contribute with us. I really like your design, thank you for sharing this. I would make the lines just a bit thinner though, what do you think? And we've been kind of unofficially using purple on our publicity so far, how to you think that would work instead of the blue? I'm curious to see what the rest of the team thinks!

ggabogarcia, 2018-10-29:
@Suzibianco  You could send me a sample of the color, or its numbering, grateful :)

stellarmagnet, 2018-10-29:
hey @ggabogarcia I really love the direction you are taking this! Regarding the purple color -- can you try something similar to the buttons here, note that it doesn't have to be exact:
[image]
[image]
- And agreed with Suzi on making the lines a little thinner. 
- How about one variation where the edges of the coral are round, instead of square?
- And a couple other font choices?

stellarmagnet, 2018-10-29:
> I described the project to an artist - this is what she made yesterday.
> 
> [image]

@jrcgarry i feel like this can work for one of our community update blog posts :)

ggabogarcia, 2018-10-29:
Hello, here are the suggestions they gave me :) I think options 2 and 5 are the best.
# 
[image]

jrcgarry, 2018-10-29:
:D
<awed>
Liking the lettering of 2 and 1.
But they're all compelling and easy on the eye.
Well done Gabriel!

founder-outbound, 2018-10-30:
I like 2, 3, and 5. Option 3 looks like the lettering from the titles in Arthur C. Clarke’s 2001, though it may not be bold enough.

 

-Craig

 

 <https://user-images.githubusercontent.com/35750428/47680215-38d62080-db94-11e8-96f4-77493895e9aa.jpg>

stellarmagnet, 2018-10-30:
@ggabogarcia I actually liked the black and grey you had in the previous design. Regarding the purple, the suggestion related to changing the accent color

Suzibianco, 2018-10-30:
Yes, agreed with Yalda. The black and grey with a colored accent seemed cleaner/clearer. But I did like the difference in line width. I prefer the "sharp"corners, and of the above, my favourite was #5.

ggabogarcia, 2018-10-31:
I made some modifications and went back to the previous color. Taking all the comments and their selections I reduced the number to two proposals. I keep leaning for the number 5 Its lines are more related to those of the isotype, besides that it tended to be better in small sizes.
# 
[image]

ggabogarcia, 2018-11-12:
I see that this problem is not very important, anyway. Thank you very much for the opportunity :)

stellarmagnet, 2018-11-12:
@ggabogarcia we apologize for not being more prompt with replying! We really appreciate your work on this logo.

The one comment I has is when I requested purple, I only meant the section you currently have shaded in as blue being purple. So then the logo would be purple and black, similar to your first set.

I also am leaning toward number 5!

Let us know if you can make this final change too :)

cagonzalez8, 2018-12-18:
I'm in love with these designs @ggabogarcia!!!