spacedecentral/Coral #45: Dust as a Astro-geologic Layer

timallard, 2018-08-02:
Defining the layer as 10m above the surface to similar density to earth soil in depth, in between will be less dense than earthen soils the static attraction well known so certain geophysical properties are not really well described or experimented with.
    It seems essential to characterize all the properties of this layer, my view that it can be a basis for filament and sintering media and additives to print with, another to find an adobe idea sans water to make it as a structural component of many uses establishing a port. 
    Consider growing hemp for the fiber and shiv for mortar based compressive loads with R-4.2/inch and fungal resistance, moisture transport, high heat resistance less spalling and enough mass, hemp ropes sailed most freight before steam, not needing wire rope where possible at scale to be considered.
    Being new to focusing on space, these are fresh idea to me, looking promising.
Cheers, tom

Suzibianco, 2018-08-02:
@timallard I'm not sure what you mean here. Is this an idea? Something to be considered? Is so, please share your thoughts in our forum so the rest of the team can join the discussion! [https://spacedecentral.net/programs/coral#discussions](https://spacedecentral.net/programs/coral#discussions )
Github is going to be used for task management, and right now we're still working on the logistics of it, so please don't add tasks right now.

timallard, 2018-08-02:
Low gravity implies "delicate" semi- compacted soils with smaller particle sizes, clays and finer that are chemically active and everyone notes the effect of static. 
    Consider looking at these as a resource of available minerals that can replace mining them the strategy.
    This to also .explore binders for what the dust may contain locally that can become an adobe clay without water as a cement replacement with fiber fillers to work with 3d printed skeletons for large buildings.
    Above the surface there is a suspended collection of very fine particles to me part of this type of low gravity soil profile that's unique so all this can supply tonnage without mining ores the idea for needed things it contains.
    A long-time GSA, AGU member, the moon soils were described to a degree yet I wasn't looking at this aspect of the dust, seemed lacking characterization for Coral and important early on, adding a definition of the layer for low gravity as a range to the process dependent on field strength if not done yet for the academic view.
    Currently working on restoration forestry and to install wet terracing for steep hillsides with very dusty volcanic & glacial till soils and no water, so I wanted to find a binder for the dirt instead of using poles and hardware cloth where it sprang up, seemed relevent to reduce mining the main reason, a mineral supply.

timallard, 2018-08-09:
This is moving to #47, Tom Mallard

closed 2018-08-09 by timallard