spacedecentral/Coral #47: Literature review on key Coral topics - Erik Kulu

erikkulu, 2018-08-04:
In response to #36 to open a new subtask and add your name to the title.

timallard, 2018-08-04:
Is my query on dust #45 suitable as a subtopic under #36 also?

timallard, 2018-08-09:
Most info on sediment contents & ore locations given, this was to visualize that first and mining covered, skimming the surface as an excavation method has value, comments on processing most agree having H2O would be mucn easier for that, so, harvesting dirt for pavers and bricks is feasible with it, otherwise sorting fines is by contrast too inefficient overall long-term a constraint.

Suzibianco, 2018-08-16:
Hi @erikkulu, how is this task going? Do you need any guidance for this? Have you found any relevant information?

Suzibianco, 2018-11-01:
[Closing this issue as we migrate to a new phase in Coral and new process of how we document issues.]

closed 2018-11-01 by Suzibianco