spacedecentral/Coral #60: Power - survival in lunar night

Suzibianco, 2018-08-16:
Find out if it is at all possible to survive the lunar night with a spacecraft with mass<10kg.

timallard, 2018-08-18:
An initial, 1st time through mind diagram of self-burial and surface options, left side of diagram: https://drive.google.com/file/d/1haFZD0QQ5FyafKluAH-iTidMz_FIvHg3/view?usp=drivesdk

Suzibianco, 2018-08-19:
Great @timallard would you like to create an outline of what would be necessary for the survival of lunar night that could be used to compare all different technologies (including the self-burial)?

timallard, 2018-08-20:
I added a section for customizing the lander, it'd make things feasible that aren't without it, mainly heating soils below it using solar-direct when day, a continuous internal parts heating sys powered by watts, radiant barrier fabrics & such we have a good chance to hone it: Diagram:
https://drive.google.com/file/d/1szgNwyYvy6mH3jJCMPLApRpM6xC0l2m3/view?usp=drivesdk

Outline: 
https://drive.google.com/file/d/1--fD3vzMj7RnGSOI6T31RgMM5pxDNESf/view?usp=drivesdk

timallard, 2018-08-26:
Latest diagram includes a radiant straw haystack node, as the final layer, from Inuit strategies: 
Diagram, 
https://drive.google.com/file/d/1lqqnvpeaw1BmkvH59hhyj8Dlj0Bhou0a/view?usp=drivesdk
Outline, 
https://drive.google.com/file/d/1L0U34vnldBa5HsfHrkBQR_18LqStganx/view?usp=drivesdk

timallard, 2018-10-08:
Regards power, this an hourly charting of panel output and doing an hour of 50-amps 3-times a day from the battery, 140w charging 400w of bus load between high current sessions: Hi,

Here's the link to the file:

https://1drv.ms/x/s!An6DguwN4uCNgZhKwBiiusg_eMzh3g

Shared from Excel for Android
https://office.com/getexcel

Suzibianco, 2018-12-20:
Closing it due to irrelevance for the time being

closed 2018-12-20 by Suzibianco