spacedecentral/Coral #69: Requirements - Subsystems

Suzibianco, 2018-08-16:
Identify the top level requirements for each of the subsystems.

founder-outbound, 2018-08-30:
Initial Task Definition document here:
https://docs.google.com/document/d/1XSfB449qn5zjmFChxTKBH7CWpSrQfRixZ1F0n575oGs/edit#

matslil, 2018-09-21:
When thinking about electronics that is classified for space and/or military use, that tends to be under ITAR rules. Do we have any requirements that we should not include any ITAR components?

The reason for me asking is that ITAR would complicate the cooperation across countries quite a lot, since you cannot send even a presentation of it freely. On the other hand, if we want any space grade electronics, it can very easily fall under ITAR. The exception would be if it is made outside US, and we are declare it properly whenever it crosses US borders.

I would actually suggest that we have a requirement saying that no item should be used that would be heavily regulated by any participating countries authority, e.g. ITAR.

Suzibianco, 2018-09-22:
This is a very tricky situation indeed... For now I think your suggestions works, however, I don't know if that will be possible as we progress.

uditkumarsahoo, 2018-09-22:
I believe we have covered it under a meta mission activity called Export Control Planning. (ref: https://docs.google.com/spreadsheets/d/1lC07ZjyZTOYCB1rvfPwLYC_lRq8tpWjqy4CPN6vKhIQ/edit#gid=953903716).
This item is expected to address these issues that will allow Coral to work globally while adhering to government regulations.