spacedecentral/Coral #70: List of partners, supliers and customers

Suzibianco, 2018-08-17:
Identify the potential partners, suppliers and customers that might be of interest to, or interested in Coral.
Refer to this document: https://docs.google.com/document/d/1jm0thyZOOJIZPHxMHr8DMsmqrsmn9wZklfqXntQ7BMY/edit

cagonzalez8, 2018-10-03:
@Suzibianco The list of the 3D Big 10, did you want me put on the google drive? If so, where would I put it?? And should I list it in this part of Github?

Suzibianco, 2018-10-05:
@cagonzalez8 , please put it  in the working research folder of pre-phase A: https://drive.google.com/drive/folders/1RbNNzGJIZqbslVens_FrwlWNJkWkIyHE

And yes, you can link it here once you've got it. Thanks!!

cagonzalez8, 2018-10-05:
*Editing in process, please disregard
Stratasys-Powder Bed Fusion
SLM-
3DSystems-
Materialise-
HP-
Protolabs-
Voxeljet-