spacedecentral/Coral #71: Existing 3D printing equipment we could use

Suzibianco, 2018-08-17:
Identify what existing 3D printing equipment we may use for initial testing - including 3D printers, feedstock, and accessories. Include suppliers and prices.

extreme-innovation, 2018-09-21:
3D PRINTER Good - Quality https://www.diabasemachines.com/product/h-series-hybrid H Series hybrid - $6,900 Technology - https://www.diabasemachines.com/technology 5 material 3d printing, cnc milling, 4th & 5th axis milling, laser scanning, touch probe leveling 300’c temperature.
Cheap
1. PRUSA I3 MK2S 3D PRINTER KIT (599$)
https://www.prusa3d.com/prusa-i3-kit
TECHNICAL SPECS (SEPTEMBER 2017 MK2S EDITION)
● 31% bigger build volume – 10500 cm3 (25 x 21 x 20 cm or 9,84 x 8,3 x 8 in)
● Open frame design for easy use
● Integrated LCD and SD card controller (8GB included)
● Up to 40% faster printing thanks to the genuine E3D V6 Full hotend
● 0,4mm nozzle (easily changeable) for 1,75 mm filament
● Layer height from 0,05 mm
● Automatic mesh bed levelling
● Automatic skew axes compensation
● Heatbed with cold corners compensation – for warpless 3D printing from any material
● Hassle free PEI print surface - no glass, no glue, no ABS juice
● Supported materials – PLA, ABS, PET, HIPS, Flex PP, Ninjaflex, Laywood, Laybrick, Nylon, Bamboofill, Bronzefill, ASA, T-Glase, Carbon-fibers enhanced filaments, Polycarbonates...
● Easy multicolor printing based on layer height
● 1 kg (2 lbs) PLA filament included
● Average power consumption 70 W (printing PLA) or 110 W (printing ABS), exterior dimensions 42 x 42 x 38 cm (16.5 x 16.5 x 15 inches), weight 6,5 kg (14 lbs), CE certification
● Specially optimized firmware for quiet printing
2. ANET E12 LARGE SCALE FULL METAL 3D PRINTER KIT(436$) https://www.3dprintersonlinestore.com/anet-e12-3d-printer
Main Features:
● 12864 LCD screen and keypad for easy offline use
● Supports 0.1 - 0.4mm printing layer thickness
● Supports TF card off-line printing, especially convenient when without the computer
● 220 x 270 x 300mm large build volume, just right for you
● More stable aluminum frame ensures your better printing experience
● Assembly is simple and suitable for DIY customers
● Working condition: 10 - 40 Deg.C, humidity 20 - 50 percent
Model: E10 Platform board: Aluminum Sheet Nozzle quantity: Single Nozzle diameter: 0.4mm Product forming size: 300 x 300 x 400mm Layer thickness: 0.1-0.4mm Memory card offline print: SD card LCD Screen: Yes Print speed: 40 - 120mm/s Supporting material: ABS,HIPS,PLA Material diameter: 1.75mm File format: G-code,STL Model supporting function: Yes XY-axis positioning accuracy: 0.015mm Z-axis positioning accuracy: 0.004mm Voltage: 110V/220V Working Power: 250W Host computer software: Cura Packing Type: unassembled packing System support: Windows, Mac Package weight: 9.052 kg
METAL 3D PRINTING - It is very costly to buy Metal 3D Printer, so it’d be better to go for get the printing done from any USA company. http://www.3deo.co/wp-content/uploads/2017/12/3DEO-Brochure.pdf Please also note the concrete printing; APIS CORE, SCULPTEO , 3D CONCRETE .

Suzibianco, 2018-09-22:
Thank you Sourav, this was in a google docs file right?

extreme-innovation, 2018-11-29:
Yes Google docs.  Inside link i guess.