spacedecentral/Coral #75: Identify all the variables of the mission

Suzibianco, 2018-08-19:
Identify the different variables that must be accounted for in the Coral mission and log them on a spreadsheet.

capsulecorplab, 2018-08-19:
I started on a serialization template that attempts to capture the space mission analysis & design (SMAD) process. Feel free to use this for reference https://gist.github.com/capsulecorplab/09ca6c698b889821f5ca0c9d26e0db99

founder-outbound, 2018-08-30:
Initial Task Definition document here:
https://docs.google.com/document/d/1SdVPVq4-VtEOfIZ3S9rEjyJHIATk__AaOG0HpRKRUbo/edit#heading=h.r7u6tnk8pst0

Engineer1119, 2018-11-29:
Notes from planning meeting:

- constraints can be somewhat quantitative. Can look to SMAD for 6 categories of constraints (cost, schedule, regulation, environment, interfaces, development).
- Astrobotics has an interface manual
- this task meant to be at a higher level for the entire mission rather than tied to any specific system.