spacedecentral/Coral #76: Raw to Printed material ratio

Suzibianco, 2018-08-21:
Research and document the ratio between raw material necessary to print a 100g object, considering the different material concentrations found in the lunar regolith.

extreme-innovation, 2018-09-21:
related links >

extreme-innovation, 2018-09-21:
https://www.3dprintingmedia.network/regolight-project-uses-sunlight-sintering-3d-print-simulated-lunar-regolith/

https://www.uni-wuerzburg.de/fileadmin/10030800/user_upload/moonbase2030/Moonbase2030.pdf

https://www.researchgate.net/publication/320929461_Sintering_of_micro-trusses_created_by_extrusion-3D-printing_of_lunar_regolith_inks

https://www.airspacemag.com/daily-planet/regolith-the-other-lunar-resource-156943194/

https://www.esa.int/gsp/ACT/coffee/2018-04-12-isru.html