spacedecentral/Coral #77: Requirements for destructive test

Suzibianco, 2018-08-21:
Research and document the minimum necessary dimensions and other conditions (shape, density, etc) to perform destructive testing on the printed object with significant results

Engineer1119, 2018-08-26:
Here is the current documentation for this task. [Materials Testing - Standards and Equipment](https://docs.google.com/document/d/1Nj5LiYxFiW3ghIH-Mp4CQJ_nJdVFYn38I4IxFEnP0T8/edit)

jrcgarry, 2018-08-28:
'Evening,
I think it fair to point out that ASTM and ISO tests (I sit on an ASTM material testing committee: C16) are designed for industrial use - and so samples tend to be quite large. There's nothing to stop one from making a miniature version of a three-point shear rig, or a tensile test rig: ASTM their ilk don't change the formulae for beam deformation etc.

On Philae and Huygens there were simple penetrometers (much like those used for soil density profiling) - but with vastly different head geometries - terrestrial standards are one thing, but as long as you can calibrate the probe/system with known materials and interpolate their behaviour, you don't need to adhere to any particular standard.
<phew>

Engineer1119, 2018-09-01:
Belatedly: thanks for pointing this out! I did a little digging on research into miniaturization of ASTM tests and the quickest way to describe the outcomes of those papers is that the viability of miniaturizing tests is a mixed bag. There's a limited amount of scalability between small samples and larger-scale samples due to various material peculiarities, and it sounds like you and your colleagues have quite a bit of first-hand experience with that. That's not even counting the fact that we'd be working with material only slightly more well known than that of Titan's surface or an asteroid.

When it comes to materials testing, what's your advice about which kind of test an ISRU demonstrator should perform? I'm leaning towards compression testing since quite a few mechanical properties can be at least inferred from it (e.g. with terrestrial concrete the tensile strength is proportional to the square root of compression strength), but I wouldn't be surprised if there is some aspect of compression testing that would make a small-scale, off-world test even more difficult than intuition would suggest.

jrcgarry, 2018-09-01:
I think that it's an interesting topic in its own right (hark! I hear an LPSC in 6 month's time!). In no particular order:

a) Yes, uniaxial compression testing is conceptually simple to perform.
No fancy attachment lugs need to be fabricated, 'merely' create a uniform cross-section object and squish it between a pair of parallel faces. The devil is in the details: the ideal test article (a cube or cylinder with smooth faces) will be, ah, a challenge. Surface wrinkles (I've tested foams at cryogenic temperatures) promote cracking, and that suggests a stage of sample preparation of some sort - or a production process which (intrinsically) leads to smooth-faced objects (!).

b) Tensile testing is somewhat harder.
Typical articles are dumbbells of revolution (think of a cylinder, with the ends flared into cones) - cunning clam-shell grips hold them. Again, smooth faces and a uniform test article are the ideal. Hard to see how either a) or b), in their canonical 'ideal' form of the test might be used on irregular and rough articles.
<but I like a challenge!>

c) Densitometry.
Knowing the uniformity of a cast/fabbed object might be of interest - nobody wants a wall with voids in it. Applicable methods might use x-rays or colimated neutrons (!). This, at least, won't demand a machined object. Fabricate a widget, pop it on a tiny rotating platform, and shine x-rays through it while using video to build a 3D volume model: density distribution calculated by inverse-methods much like CT scanning.
<https://www.medgadget.com/2013/01/tiny-x-ray-source-uses-piezoelectricity-to-generate-radiation.html> O_O

d) Thermal conductivity. 
C177 is for the birds. We'll have a non-pancake like widget, and no way to ensure perfect heat transfer into it.
But - we'll have the best insulation possible - so a rough beam might be warmed at one end (apply with optical/IR rather than trying to stick a heater onto a rough ceramic surface), and then IR radiometry might monitor the rate at which heat conducts along the beam. If you know the average density and specific heat, you can calculate average thermal conductivity.

e) Porosity.
Mmm. Traditional pycnometry might be viable. A tiny hermetic volume is needed and whiff or two of helium. Only measures connected pores - not wrong.

f) Bending/shear.
Same problems of fabricating smooth faced samples as uniaxial tests. I'd skip it.

g) Our ol' friend penetrometry: https://www.sciencedirect.com/science/article/pii/S0032063300001690
Avoid. Requires calibration against known analogues. Don't see the point (oh, sorry).

h) Oddball properties:
* albedo (glasses made from JSC are black: https://isru.msfc.nasa.gov/lib/Documents/PDF%20Files/ISRU-ISFR-MRS-NR.pdf) and emissivity would be fun to know for accurate thermal budgets. 

* hardness, simply is the slope of your stress/strain curve in compression

* coefficient of thermal expansion: small widget = small strains, tricky to measure even with an interferometer (may be little backscattered to create a 'test' light beam with.

* magnetic / dielectric properties? I can imagine that a cast/sintered object may have mu and epsilon that differ from raw regolithhttps://isru.msfc.nasa.gov/lib/Documents/PDF%20Files/ISRU-ISFR-MRS-NR.pdf

* volatile content in raw state? Fabrication may involve significant heat - so having a gas analyzer (a teeny tiny quadrupole? Ptolemy on Beagle2 was a *small* ion-trap for just that) to identify and quantify evolved volatiles might be nice to know: are the best bricks made from regoliths poor in volatiles, if so, which regoliths should be avoided? etc.

Summary:
Compressive strength, and ideally a stress/strain curve, would be of greatest interest to a would-be builder of lunar infrastructure. Major obstacle is in preparing a sample such that heterogeneity of specimen is minimized.

After that, density mapping and thermal properties seem to be both informative and fairly tractable.

timallard, 2018-09-02:
Consider that strength of materials curves is the goal vs duplicating machines in miniature when stresses may not scale well.

Taking that approach and avoiding the yoke-and-press structure for c-clamp style, then, a glance at regolith shows we need to sort it and examine categories of materials in tests, as-is by all comments needs a binder (or using heat pressing at far lower pressures to gain strength), with this the main testing results are by using a tiny foundry and disk samples up to 7.5mm thick by 12mm in diameter as a machine able to return good data on many combos of materials.

Cast iron cups can use induction heating, the solar-direct sintering demo uses a thermal fluid as a day input, it's the heater for the lander and can cast, sinter, stamp & hot-forge demo at that size within reason on weight & power.

Automating the "assembly line" uses std opc, can be IIoT, press foot & anvil choices on rotators, cups for stamping & forging as well. 

Cast glass & ceramics have a lot of combos by having sorted feedstock to test from.

The test press solenoid driven, cap discharge, a single point to bend to breaking, a valley anvil tests beam span, a chisel press foot for this and shear test. 

Optics & interferometry augment the other sensors to create a decent materials library of many variation of especially the agglutinates, upland nickel is 5% of them so by continuous output and many tests enough for a thin coin can be gathered.

Goal is for no waste, the dust may lend itself to using a plasma to form droplets as a feed, solar-direct light can be treated to be laser quality & power for a welding demo, direct sintering is too awkward with optics, the sun follower uses thermal fluid to transfer heat to cups, allows flex hoses to the foundry.

Sequence is pick-n-place, hoppers with sorted regolith move above cups, eject to other steps then test station, composition & properties recorded, breakage can be recycled.

Results are calibrated to astm by returned sample processing for an end goal. Static charge use is an open field, agglutinates sort by reflective index, beyond magnetic using N.Tesla valvular conduits to sort is non-mechanical and gives 6-7 bins initially by vortex power, the bins can then be sorted again by a conduit and other means avoiding it getting into moving parts. 

Please consider this overall concept as putting down a wide scope of engineering metrics of possible product venues at scale by testing tiny and calibrating.

Engineer1119, 2018-09-02:
Yes, generating some stress-strain curves would be a great deliverable set of data for the payload. 7.5mm x 12mm is oddly specific, are you basing those dimensions off of something? I don't have a problem with those numbers, I'm just curious.

Some of your proposed systems and methods may be difficult to scale down (at face value) but they sound worth looking into. Please upload any research you find on those to the Drive and record them in the library.

Good thinking about follow-on missions to retrieve samples that allow calibration in a more robust lab setting. Some of the extended or more-complex processes you mentioned may have to take place on a follow-on mission as well. It reminds me of the NASA architecture for a robotic Mars sample return mission.

This might be my personal bias towards structural engineering and testing considerations, but I'm starting to wonder if we can start by defining the minimum useful test article size and then work backward/upward from there to define the required mass, power, and dimensions of the test rig, then further backward/ upward to the required properties of the other systems and subsystems that support it.

timallard, 2018-09-03:
22mm, apology, missed the 12mm typo, size of a nickel.

Last night I did scaled sketches of press foot and anvil shapes to gain direct beam span deflection, shear, bending from a free end gives tension & compression and finally a direct compression test with a pin punch shape that crushes an area in the center.

This all fits about 3"x4" and the solenoids attached, moving disks from the cups nearby upon eject, weight so far 6-7 lbs with sorting a guess.

This setup with sensing should provide enough metrics to be a broad foundation of what use these materials can be for, from solidstate electronics to the structural needs of buildings on a first flight, it may become a standard if vetted being so small.

Seemed worth a prototype on the free-end bend test on glass with optics, pretty sure that can give useful results to move forward on.

timallard, 2018-09-11:
We can satisfy this ASTM flat glass spec, non-destructive, first glance there are enough others to baseline a minimum of 5 certifiable glass or ceramic tests, not considering metal tests this flight: https://www.astm.org/Standards/C1279.htm