spacedecentral/Coral #78: Research rides to the moon

Suzibianco, 2018-08-21:
Research potential rides to the moon, considering their landing sites, expected launch date, landing vehicle capabilities, ride sharing cost, etc.

Engineer1119, 2018-09-02:
I uploaded the payload user guides for Astrobotic and PT Scientists and a screenshot of Moon Express' MX-1 lander capabilities  to the "mission engineering" folder. Team Indus and SpaceIL don't strongly signal their intent to fly payloads by other entities beyond their current partners.

Suzibianco, 2018-09-11:
@Tamalee07 and @founder-outbound please coordinate and work on this together - rather than having two people working in parallel in the same task 😃

erikkulu, 2018-09-15:
Moon Express can likely be removed from this list unfortunately https://twitter.com/JoeSolarsystem/status/1025506710700605440

founder-outbound, 2018-09-15:
That’s very disappointing. ☹