spacedecentral/Coral #81: Surface site reference database

Suzibianco, 2018-08-22:
Generate a reference database on sample properties from all surface sites (Apollo 12, 14, 15, 16 & 17; Luna).
*Regolith physical properties (% regolith vs rocks; size range of particles, etc.)
*Regolith mineralogy and constituents (Content percentage).
*Regional access and terrain
*Landing engine impacts/interaction on surface materials
*Stability of regolith surface for potential structures

Draft Example:

[image]

timallard, 2018-09-14:
A regolith density equation adaptable to highland or mare soils, porosity from agglutinates a percentage, the other two terms are basic specific-gravity then depth: https://drive.google.com/file/d/1KobDfKSZukTNKKuHQcfWs6FwGzG8u7eN/view?usp=drivesdk

jrcgarry, 2018-09-14:
Tom,
Do you have a reference for that? I'm struggling to parse it.

timallard, 2018-09-18:
@jrcgarry I haven't fount the log function, yet, for surface you can use estimated values from just the average density for the type from the first integral, forget the term for depth and use porosity percent by type, doing that I came up with 2-1/3 g/cc two ways for our landing site. If anyone finds the log function we'll add it, keeping an eye open.

jrcgarry, 2018-09-18:
Tom,
I meant, do you have a reference for where this equation came from?
(textbook, paper, etc.)

timallard, 2018-09-19:
@jrcgarry It's my equation as an estimation tool, I took the limits from Chapter 6, Table 6.8 on Mare vs Highland soils, still no dice on the log function. This gave me a 3-1/3 cc shot per coin of as-is regolith for Mare, it's more for Highland from that being more porous than Mare by quite a lot, 59% voids vs 26%.

This also implies less compaction for adobes from Mare soils, heat pressing will be faster.