spacedecentral/Coral #83: Create a task template

Suzibianco, 2018-08-22:
Create a template for describing tasks, with a common structure that can be used for all tasks. Refer to this article: 
https://medium.com/nycplanninglabs/writing-a-proper-github-issue-97427d62a20f

founder-outbound, 2018-08-23:
Initial Proposed Template

Title of Task
          Endeavor to be succinct but accurate with the title. Always ask yourself: "Is this a vague title, or does it give a unique and understandable description that is immediately understandable?"

Description of Task
          A much more in-depth accounting of what, who, and how for the task.

                       What: The Purpose of the task, the reason it exists. This can be of considerable detail to effectively explain to persons interested in the task.
                       Who: Name he principal responsible party and any other persons on the task.
                       How: Intended strategies to address any subtasks. Multiple strategies can be listed.

Timetable
          Establish an end-to-end schedule of events and milestones to accomplish the task and subtasks.

Suzibianco, 2018-08-23:
@founder-outbound I'd say this is a good start. what do you think @capsulecorplab ?

capsulecorplab, 2018-08-27:
LGTM

founder-outbound, 2018-08-28:
I am unfamiliar with that acronym.

founder-outbound, 2018-08-28:
I've tried to clean up the comment-version of the template as a Google Doc, here:
https://docs.google.com/document/d/1wwmf7I_G3bfuRv5qLFvuPz9CYLj54UToJ85NThvCNVI/edit

founder-outbound, 2018-08-28:
Didn't mean to close it...

capsulecorplab, 2018-08-28:
LGTM = "Looks Good To Me", primarily used in the developer community to express approval or favorable feedback :slightly_smiling_face:

Engineer1119, 2018-09-01:
This looks great.

stellarmagnet, 2018-09-09:
I would also add something like this (clarifying to the person completing the task the high level criteria for the work to be accepted) and how many times they can update their work if it was rejected

## Acceptance Criteria
A correct submission will:
- criteria 1
- criteria 2
- criteria 3
### Revisions
We will review up to 2 revisions of the submitted work

Engineer1119, 2018-11-26:
I added the content of Craig's google doc + Yalda's revision item to [this document](https://docs.google.com/document/d/1wbTkeQEVOcC9sX_MwdGwdWCxvkcP7yS5kQI6c3oOWDY/edit#) linked primarily to task #52 (creating the broader Contributing Guide). Well done!

closed 2018-11-26 by Engineer1119