spacedecentral/Coral #86: Simulant testing specs

Suzibianco, 2018-08-30:
Research/find out the technical specs of the 3D printers that were used in the simulant tests. 
- Type of printer
- Composition of simulant
- Printer head specs
- Process method

pratik1094, 2018-09-12:
There is a research conducted by Loughborough University,UK. They used Realizer SLM™ 100 for printing different geometrical shapes using Lunar regolith simulant (JSC-1A).

Printer Specifications-
Laser Near-IR Ytterbium Fibre Laser
Power (W)-10-50W
Wavelength (nm)-1,070-1,090
Diameter of laser spot (micrometer)-100-300
Point distance (micrometer)-20-300
Hatching space (micrometer)–(Overlap %) 210-300 (35-0%)
Exposure time (microsec)-1,000
Thickness of powder layer (micrometer)-100-350
Velocity (m/s)-Point distance/Exposure time
Environment-Argon atmosphere
Temperature of substrate (°C)-200
Substrate material-Mild steel with layer of regolith clad

The process which they used is Selective Laser Melting.Its a type of Powder bed fusion method which involves fusing granular particles using laser.
As far as findings are concerned, the dimensional accuracy of 99.8 per cent were observed in the fabricated samples.
Some relevant link-
https://www.youtube.com/watch?v=OIBRIM_P2X0

jrcgarry, 2018-09-12:
The link to the author's work:
https://www.researchgate.net/publication/277312590_3D_printing_with_moondust
<click on 'Download'>

TLDR: Several other groups have looked at laser sintering.
Balla et al. 2010
https://ntrs.nasa.gov/search.jsp?R=20110000850&hterms=laser+sintering+regolith&qs=N%3D0%26Ntk%3DAll%26Ntx%3Dmode%2Bmatchallany%26Ntt%3Dlaser%2Bsintering%2Bregolith

Colley, 2015.
http://www.acser-archive.unsw.edu.au/oemf2015/slides/DAY2-11.pdf

Indyk & Benaroya, 2017.
https://www.researchgate.net/publication/319951295_A_structural_assessment_of_unrefined_sintered_lunar_regolith

Challenges, as I recall, are ensuring an even bed temperature (you want to minimize the work needed from the laser), and how one applies a consistent layer of regolith for each pass of the laser is a knotty problem.

founder-outbound, 2018-09-15:
In the hustle and bustle of the past week, I missed this information. The below results are incredibly exciting to me! On my day job, we used powdered Inconel to make a high-temperature mesh using laser sintering, and it worked beautifully, both thermally and structurally. The required performance for a structural building-block will be much less demanding, and if we can use laser sintering on real regolith like with the results from Loughborough, it could be the key to success. No water needed, just plentiful electrical power and a the ability to change the relative position of the sintering point. Oh, and of course a means of regolith collection and material characterization.

 

https://www.youtube.com/watch?v=OIBRIM_P2X0

timallard, 2018-09-17:
Personally it's not that different than heat pressing as-is regolith with the complication of a laser, ASTM uses heat pressing sediments as a method on dry sediments so is preferred for that reason also for bulk processing, pressing is way faster. The coinmaker has a 5-6 tests including heat-pressed as-is regolith to fully melted, it';s really smelting and by controlling temps well one can let that sort-by-slag like smelting to combos you miss in a full melt.

Also, this can't be done inside the lander another constraint, dust-free mfg is for a reason around electronics, suggest any tester is autonomous digging under the lander, this also satisfies a list of req's part of why to do it, mobility, autonomy via comm two biggies.