spacedecentral/Coral #89: Migrate knowledge base to github

capsulecorplab, 2018-09-05:
This is a high-level task for outlining and discussing strategies for migrating Coral's knowledge-base from google drive to github. Purpose for this migration will be to better facilitate task management using github’s built-in source control system, as tasks grow in complexity.

Exact implementation will entail use of a markup language supported by github. Primary candidate in consideration for this implementation is [AsciiDoc] (https://asciidoctor.org/docs/asciidoc-writers-guide/) as it's the recommended markup language for technical writing, and a more [preferred alternative to markdown](http://ericholscher.com/blog/2016/mar/15/dont-use-markdown-for-technical-docs/).
Custom templates may also be needed depending on project management decisions pertaining to the knowledge-base ontology.

capsulecorplab, 2018-09-06:
@cagonzalez8, this may be a bit of a monster of a task, but would this be something of interest to you? 

I'm labeling this as an icebox item for now, but the sooner we adopt such a solution, the less work I suspect it'll be having to migrate any existing work into the "Github time-capsule". @founder-outbound @Suzibianco @Engineer1119 your feedback is welcome as always.

Suzibianco, 2018-09-06:
Seems good to me, I am beginning to understand the urgency 😅