spacedecentral/Coral #90: Define the full con ops & create diagram

stellarmagnet, 2018-09-11:


uditkumarsahoo, 2018-09-15:
I have considered the following requirements in the [requirements document](https://docs.google.com/document/d/1kpqc1co7viuFuppOO8iwucS9kJ2HNmybD3LC2ruH-4o/edit?usp=sharing) and added it to the [meeting agenda sheet](https://docs.google.com/spreadsheets/d/1lC07ZjyZTOYCB1rvfPwLYC_lRq8tpWjqy4CPN6vKhIQ/edit?usp=sharing).

Mission Level Requirements: 1.3, 1.5, 1.6, 1.7, 2.1.1, 2.3.1, 2.3.2, 3.1
Requirements/Activities: 2.3.3.1, 2.3.3.2, 2.3.5, 2.3.6, 3.8, 3.8.1, 3.8.2

The other requirements may not be applicable/redundant at this point.

Please do feel free to comment on the same.

uditkumarsahoo, 2018-09-15:
I just in-correctly closed the issue. Re-opening the same.

Engineer1119, 2018-11-25:
for posterity: sheet created and populated by Patrick, Yalda, Suzi, Sean, Udit, Craig, Carlos, Mats, Brayden, Marc, Sourav, Tom, Tamalee. Reference requirements document created by Andy.

closed 2018-11-25 by Engineer1119