spacedecentral/Coral #91: Trade study - Sensor options

stellarmagnet, 2018-09-11:
We need to be able to analyze the composition of the lunar regolith before processing, during the collection process.

- Type
- Mass
- Power

jrcgarry, 2018-09-11:
Yalda:

From memory of what has been flown on similar landers I've known:

a) Mechanical bearing strength
(PEN on Huygens: a hemispherical penetrator attached to the underside - a PZT force-transducer allowed for discrimination of grain sizes)
(MUPUS on Philae: a hammered penetrometer - an electromagnetic 'hammer' was to percusively drive an instrumented spear into the comet)

Mass: will scale (roughly) with the size of the specimen being poked.
Power scales with speed of compression
Complexity: low
Cross-coupling: might annoy any seismometers on the lander

b) Neutron densitometry
Allows for an estimate of bulk density (Apollo 17)
Mass: independent of specimen size
Power: almost none
Complexity: low
Cross-coupling: might annoy any particle payloads on the lander - choose energies carefully!

c) Electrical permittivity
May allow for an estimation of water ice content (at the surface!?) Used on [Philae](https://tel.archives-ouvertes.fr/tel-01496107/document), as part of the SESAME package.
Mass: almost none (it's a tuned LCR circuit - and you watch the Q of the cavity)
Power: some - not a lot (Watts)
Complexity: modest (you've got to clean the cavity or somehow slide parallel plates into the specimen.

d) Thermal properties
Higher density and lower porosity leads to higher thermal conductivity. Specific heat scales with density.
Mass: almost none (similar problems to electrical permittivity)
Power: some - maybe 10s of Watts
Complexity: low (Philae again: [https://www.researchgate.net/publication/225004358_MUPUS_-_The_Philae_Thermal_Properties_Probe](url)

e) Volatile content
Outgassing during heating is of great interest. QCMs (quartz crystal microbalances) might be the only viable methods apart from mass-spectrometry. 
Mass: almost none (the electronics box will dominate the sensor by 4 orders of mag)
Power: low
Complexity: considerable

f) Tribology and electrostatic phenomena
A host of questions surround the behaviour of regolith grains when moved, illuminated, or charged.
Outside our immediate concerns.

I'm happy to write the above up as a Research Note for trade-off input. Might add to the list as the day wears on.

And now some fun links:
[http://adsabs.harvard.edu/abs/2018arXiv180105754M](url)
[https://agupubs.onlinelibrary.wiley.com/doi/full/10.1002/2014JE004648](url)
[https://agupubs.onlinelibrary.wiley.com/doi/full/10.1029/2009JA014559](url)
[http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.116.2987&rep=rep1&type=pdf](url)

stellarmagnet, 2018-09-12:
@jrcgarry thank you for sharing all of these details! A Research Note for this sounds great, please add it to the 'Working Research' folder here (still needs to be a bit organized):
https://drive.google.com/drive/folders/1RbNNzGJIZqbslVens_FrwlWNJkWkIyHE

jrcgarry, 2018-09-12:
>  A Research Note for this sounds great,

I'll enjoy that far too much.
Thanks!

-JG

jrcgarry, 2018-09-24:
<puff pant>
There we go. A first cut at the document of How One Measures Regolith.

I'll throw in some more photographs of flown items, and make an attempt to gauge power & mass in order-of-magnitude scales based on present SoA.

[https://drive.google.com/open?id=1qJMMwqnQ-9Qg1Z1AAWGWu6TtJg2acIzUkGMAG-Kn3tM](url)
Comments/critiques welcome.

jrcgarry, 2018-10-16:
Hi Yalda,

As the #91 document seems to have stabilized does it get handed around for review?

Cheers;

Suzibianco, 2018-11-01:
@jrcgarry yes, I will pass it around to a few other people in the team. Thanks!

Engineer1119, 2018-11-26:
@jrcgarry I fully read your research note and love how thorough you are with it. Thank you for putting this all together.

Flying even 1 of each of the instruments you discuss sounds like an entire mission on its own 😟 which means down-selection at some TBD point that I need to define. In that sense we probably won't formally conclude the full trade study right away, but I'll do my best to keep an eye out for when we need to fill in more areas that you've tagged as in-progress.

jrcgarry, 2018-11-26:
@Engineer1119 Rest assured, the smorgasbord of sensors and techniques presented is simply there to kick ideas around.
In all likelihood, one (or at most, two) methods might be used to infer some quality of the regolith prior to its processing. And depending on the desired accuracy and thus reliability of their inference, these could be little more than a pair of wires in the fabricating stage.
<mumble: whatever *that* looks like>
Thanks for the read.