spacedecentral/Coral #99: define file naming convention

Engineer1119, 2018-11-29:
(Patrick to fully fill this out)

capsulecorplab, 2018-12-07:
Open to alternative suggestions, but one idea for a naming convention was a cross between the document type and the first 8 characters of a version 1 universally unique id (UUID) hash, e.g., `SOP-ed6a94ca` is an instance of a standard operating procedure. The document itself could optionally contain the full UUID, ed6a94ca-fa45-11e8-8eb2-f2801f1b9fd1, as an artifact that could easily be parsed (depending on the document format) and used for other identification purposes if need be.

Engineer1119, 2018-12-09:
> Open to alternative suggestions, but one idea for a naming convention was a cross between the document type and the first 8 characters of a version 1 universally unique id (UUID) hash, e.g., `SOP-ed6a94ca` is an instance of a standard operating procedure. The document itself could optionally contain its full UUID, e.g., `ed6a94ca-fa45-11e8-8eb2-f2801f1b9fd1`, for other identification purposes.

That would certainly help with any future efforts to make parsing our files easier for a machine, but a UUID hash wouldn't really help us poor humans out that much.

Engineer1119, 2018-12-09:
I started putting together a list of file naming rules [here](https://docs.google.com/document/d/1gTMqg0xMh8Jss0smt9JTIPiGFINcj2fVYstDzoquqBw/edit). Please let me know what you think by commenting in the doc or on Github.